engineertype/Controleo3 #14: Parametric Profiles

rafl, 2020-12-01:
I love my Controleo3, but find the process of creating and refining profiles very frustrating.

For reflow soldering profiles I usually start out with the solder manufacturer's recommendations, but almost inevitably there'll be some tweaking I'll want to do to make the profile work well with my particular oven and the particular boards I want to make. That process typically involves just changing temperatures by a few degrees or times by a few seconds, but currently requires removing and inserting the SD card many times to get to the right recipe.

I also use my Controleo3 for applications other than reflow soldering, where I typically just want to keep a one-off part at a given temperature for a given time, but both time and temperature tend to vary from part to part.

In both of those situations I'd find it really useful if the Controlleo3 firmware supported adjusting certain profile parameters on the fly, without having to plug the SD card into a different device to make a tiny change.

To address that frustration, I'd like to propose the idea of profile parameters.

Profile parameters would be declared within profiles using a new token, for example: `Param soak_duration default 100s`. Once declared, other profile commands should then be able to reference those parameters, for example: `Ramp temperature to 165C in soak_duration`.

When running a profile using parameters, the user interface would prompt the user to specify a value for each parameter by adjusting the profile default, and then execute the profile with the given values effectively substituted within the rest of the profile.

I'm not proposing anything more complicated than literal substitution. While I could see potential uses for an expression language that can represent `Ramp temperature to 165C in soak_duration + 10s` and similar things, that seems out of scope right now and could be added later. Similarly, while saving a set of parameter as a named profile or persisting past parameter values of a profile seems like it might be a useful feature if profile parameters were to become a thing, I'm not proposing any of those things at this time.

I'd love to hear people's thoughts on that idea, if the maintainers would be open to such a feature, and if so, if they'd have any thoughts on the specifics of the syntax and semantics of such a new profile language feature.

Thanks!

rafl, 2020-12-01:
Oh, and I should've mentioned that I'd be more than happy to implement such a feature. I was just hoping to get some input on the idea and the proposed syntax before jumping into the implementation.

engineertype, 2020-12-01:
Most users experiment with their profiles for a while, tweaking values so that they work with their paste and reflow oven.  Once this is done it is just a question of running the same profile over and over again.

I appreciate your enthusiasm on this.  My design philosophy for profiles for Controleo3 was to make them powerful and flexible - but also keep them simple and easy to use.  I don't believe profile parameters align with this.  Thanks!