engineertype/Controleo3 #15: Door held close feature

bobjunga, 2021-05-12:
I just built my reflow oven. The kit rocks! Well done. Thanks.

I am planning to add a feature to the firmware and would appreciate your feedback. Also, if if is a feature you would be interested in adding I will do it in the spirit of your code and submit a PR when done. Otherwise I might play around for myself while I am in there and just keep the fork for myself (and anyone else that might be interested). 

I would add a door lock mode. I will make a servo arm with another arm out in front. it would be clear enough not to interfere in the existing operation but then if you drive the servo past the closed position, that arm would engage the metal door plate on the other side and hold it firm against the oven. I am not sure if the existing servo will have sufficient range of motion but I assume I can one one that does if needed. 

Why I am doing this...

My oven is a Black & Decker TO1760SS in which the door overlaps the oven frame and the curve of the glass door does not match the curve of the frame very well so adding the liner leaves big gaps. My solution was to lay a bead of the red silicone, wait 10-20 minutes, put a film of motor oil around the door where it would touch the bead and lightly pulse the door closed to mold the bead to the shape of the door. This worked pretty well but the door does not close tight on its own. It does not take much force to get it to make a tight seal, though.

--BobG

engineertype, 2021-05-13:
I don't have anyone else asking for this feature.  In the spirit of keeping things simple I won't add your change to the main code base.  I think the best approach is to add a linear actuator (solenoid) controlled by one of the outputs.  If you name this output "cooling fan" then the profile commands to turn the "cooling fan" on and off can activate the actuator.  No need for any software changes!

closed 2021-05-13 by engineertype

bobjunga, 2021-05-14:
Interesting. Thanks for the info.

--BobG