engineertype/Controleo3 #17: Build tip 100k grounding doesn't make sense as error cause

gwideman, 2021-09-12:
In Step 14 of the Controleo3 build guide, there's a Tip  _"in some cases static builds up on the thermocouple so the MAX31856 IC reports an over/under voltage error and Controleo3 won't display the oven's temperature. Touching the tip of the thermocouple with your finger removes the static and the error for the duration of the reflow session. [...] However, grounding the thermocouple fixes this as explained here [link to NI explanation]. "_

But I don't think this makes sense.  The NI document focuses on a thermocouple connected to differential inputs of an opamp or differential amplifier, which are very high impedance, but source or sink some bias current, and thereby might charge the thermocouple assembly.

But the Controleo3 uses MAX31856 that provides a bias terminal to deliberately set a common-mode voltage on the thermocouple wires. (And the PCB does use that bias terminal.)  So static charge cannot build up on the thermocouple per se, in the manner discussed by the NI document. 

Even if static could build up on the thermocouple as described in the NI doc (ie: not applicable to the MAX31856), then providing a path from the Controleo3 ground to the chassis ground wouldn't cure it, because that would not drain charge from the thermocouple.

So, if an error condition is indeed detected and reported by the Controleo3 (is this an established fact?), it seems to me it's not related to the thermocouple acquiring a static charge per se.  

But if could be due to some aspect of the entire Controleo3 floating relative to the oven chassis.  It would be good to elucidate what exactly is going on there, and determine whether the Controleo3's ground rail should be:

1. Not grounded, or
2. Provided with a high-resistance path to chassis ground, or
3. Grounded direct to chassis ground.

I would expect that option 3, which puts Controleo3 ground at the same potential as the chassis, would make best use of the chassis to shield against possible noise pickup by Controleo3 (or the thermocouple wiring). However, I don't know whether there's some other obtuse mechanism at work here, given that inside the oven, signal wires might be routed near the AC wiring for the heaters.

engineertype, 2021-09-25:
Thanks for the detailed write-up.  This doesn't happen very often, so troubleshooting has proven to be difficult.  We had one oven exhibiting the problem, and we tested #1, #2 and #3 above.  #1 gave a "thermocouple under/over voltage error".  #3 made the thermocouple read 20C too high.  For #2 we first tried a 10k resistor to chassis ground.  This fixed the under/over voltage problem but again caused a high reading.  We found the best was a 100k resistor to chassis ground - and this is what the guide recommends.

Our sample size is "one", and some customers who experienced the problem reported that a resistor fixed their issue.  I agree it would be good to understand exactly what is causing the problem, but in the mean time the fix appears to be solid.  Also, it is good to frame this issue as affecting around 1% of builds.

gwideman, 2021-10-06:
Sorry for delay in replying. I've now built up my unit and done some testing. So far as I can tell, it makes do difference to the temperature reading whether the negative rail of the Controleo3 is grounded to the oven chassis (in turn connected to wall outlet ground) or not, or connected with a resistor.  

That being the case, I have grounded the negative rail to the oven chassis, because this prevents the Controleo3 (and thermocouple) from floating at half the AC supply voltage (ie: near 60VAC).  This would occur because the output of  unpolarized 2-pin AC adapters (including this one) sits at mid AC supply, as there's about equal capacitance from each of the AC line inputs to the DC output.  This voltage is at a quite high impedance, so would delier only a tiny current, but there's no point letting that voltage be present on the thermocouple, where the tip could short to the chassis if touched to the tray, or worse yet touch a sensitive component on a PCB that's resting on the grounded tray. 

Getting back to the unit you saw that benefitted from the 100k resistor from negative rail to ground (but not direct connection or 10k), I would say that almost certainly indicates something else is wrong.  If Controleo3+thermocouple is essentially isolated from the rest of the universe, adding a single current path (like a resistor or wire) to ground should have no effect on the temperature reading, as there's no complete circuit.  So I'm thinking that there was actually already some unplanned conduction path from the chassis to the thermocouple -- perhaps chaffed-through insulation or somesuch.   

As I mentioned before, I'm pretty certain that the explanation could not be as discussed in the NI docs, as you're using the bias output to set the baseline voltage on the thermocouple, so it can't float out of range.  Unless, of course, on that one unit, the bias pin of the MAX31856 wasn't soldered or some similar mishap.

Anyhow, just some thoughts from having nosed around a bit further with an actual unit.